espressif/esp-video-components #5: master: Cannot build if CONFIG_ESP_ENABLE_ISP is disabled.

jitenshap, 2024-09-04:
Hello,

I have high expectations for the ISP support in this project. I recently cloned the latest master branch and integrated it into my project. I enabled the "Enable ISP based Video Device" option (CONFIG_ESP_ENABLE_ISP) and attempted to build my project.

However, the call to 
int fc = open("/dev/video0", O_RDONLY);
which was successful with the release version distributed via the Component Manager, now fails.

I haven't investigated the issue in detail yet, but when I disabled the "Enable ISP based Video Device" option and rebuilt the project, the build failed. The error messages suggest that the ISP-related code added to esp_video_csi_device.c is not excluded by the #define statement, resulting in errors such as "implicit declaration of function 'esp_video_isp_enum_format'."

While I am concerned that the device cannot be opened when the "Enable ISP based Video Device" option is enabled, my immediate request is to ensure that the project can be built successfully when this option is disabled.

Thank you for your attention to this issue.

Best regards,

donghengqaz, 2024-09-05:
Hi @jitenshap thanks for your feedback, the basic ISP feature is ready recently. Please open the option "Enable ISP based Video Device" although you use integrates ISP, the ISP video device will pass P4's ISP and directly output sensor's pixel data.

jitenshap, 2024-09-05:
Thank you for your reply.
The config I wrote was incorrect. instead of CONFIG_ESP_ENABLE_ISP
CONFIG_ESP_VIDEO_ENABLE_ISP.

I understood that in principle CONFIG_ESP_VIDEO_ENABLE_ISP should be enabled.
I look forward to you adding examples for the component that include ioctl to configure new ISP features.
Currently I'm looking at the greenish sensor raw data.

Thank you.

closed 2024-09-05 by jitenshap