espressif/esp-video-components #7: MIPI CSI camera cannot be opened in master branch

jitenshap, 2024-09-16:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Open MIPI CSI camera and start capturing video.

### Actual behavior (suspected bug)

Failed to open camera device.
`fd = open("/dev/video0", O_RDONLY);`

### Error logs or terminal output

```shell
I (1614) main_task: Calling app_main()
E (1614) example: failed to open device
E (1624) example: Camera capture stream failed with error 0xffffffff
I (1624) main_task: Returned from app_main()
```

### Steps to reproduce the behavior

Build examples/capture_stream on the master branch.
Then flash to ESP32-P4-Function-EV-Board v1.4 and run.

### Project release version

master

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

PowerShell

### Additional context

IDF Version: v5.4-dev-3023-g3c99557eee

WangYuxin-esp, 2024-09-23:
@jitenshap May I know which camera sensor you are using?

jitenshap, 2024-09-24:
Thank you for reply.
I using camera is SC2336 bundled ESP32-P4-Function-EV-Board.
I found that CAMERA_SC2336 is Disabled in SDKConfig.
After enabling this, I was able to open the camera device. 
Sorry for the mistake.

closed 2024-12-06 by wujiangang