espressif/esp-video-components #10: Support for SC132GS sensors

simonzsy, 2024-11-19:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

High resolution global shutter sensor 

### Use cases

For machine vision and video data collection

WangYuxin-esp, 2024-12-06:
Thank you for raising the request. We are evaluating support for SC035HGS and it may take some time to test the final image quality.

WangYuxin-esp, 2026-08-28:
The driver for SC132GS has been released on GitHub. Thank you for your interest.

closed 2026-08-28 by WangYuxin-esp