espressif/esp-video-components #12: 参数及类型错误，无法

haolsh, 2024-11-30:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

[image]

[image]

### Actual behavior (suspected bug)

master分支

### Steps to reproduce the behavior

esp-video-component 中的esp_video  以及esp_ipa 函数参数错误

### Project release version

espidf-5.4.0

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

ubuntu22.04

### Shell

ZSH

wujiangang, 2024-12-06:
@WangYuxin-esp Please check about this.

donghengqaz, 2024-12-09:
Hi @haolsh, thanks for your feedback, this compiling error is fixed in commit https://github.com/espressif/esp-video-components/commit/b5fd2933327118f739cf1e0f0ec2c4f82d08f887, PTAL, or you can also try the master branch.

wujiangang, 2024-12-09:
It is generally recommended to use the version available on the [ESP-Registry](https://components.espressif.com/). However, using the latest version and reporting any issues is also appreciated.

closed 2024-12-09 by wujiangang