espressif/esp-video-components #18: OV9281 support

nicklasb, 2025-01-04:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

I would like to add MIPI support for the OV9281. 
If you are not planning to, is there any way for me to add support for a specific sensor myself? 
Do you use any scripts or tools to generate the config byte streams or something?

### Use cases

The OV9281 is a grayscale sensor that great for high-speed and low light applications, it seems like this is a use case that is not well represented among the other sensors?

WangYuxin-esp, 2025-01-09:
@nicklasb Hi, Thanks for your request. [Adding a new RAW sensor driver](https://github.com/espressif/esp-video-components/tree/master/esp_cam_sensor#steps-to-add-a-new-camera-sensor-driver) usually requires a lot of tuning work and requires the sensor manufacturer's technical support for related development work. We have reviewed the technical parameters of the OV9281 sensor and are considering adding a similar camera driver in the future. I will reply to you here after this development work is completed.

nicklasb, 2025-01-09:
> @nicklasb Hi, Thanks for your request. [Adding a new RAW sensor driver](https://github.com/espressif/esp-video-components/tree/master/esp_cam_sensor#steps-to-add-a-new-camera-sensor-driver) usually [...]

Cool. What is the similar camera you are looking at?
I am always looking for alternatives..

WangYuxin-esp, 2025-01-09:
Maybe we will try it on the sc035hgs sensor(480p, global shutter, 120fps...), since we have already added sc035hgs driver. If it works well, we will consider adding a 720p global shutter sensor.

nicklasb, 2025-01-09:
Ok, It *sort of* meets my most basic requirements, however, it doesn't seem available anywhere?

nicklasb, 2025-01-09:
Understood. But again, the sc035hgs doesn't seem available anywhere and have very few sales and no reviews i can find?

WangYuxin-esp, 2025-01-09:
> Understood. But again, the sc035hgs doesn't seem available anywhere and have very few sales and no reviews i can find?

Perhaps after tuning, we can provide a link for everyone to buy [it](https://www.smartsenstech.com/en/products_list).

nicklasb, 2025-01-09:
Ok. I am really trying not to be negative, but it does not seem like sc035hgs (which has no datasheet in the link you provide) is very widely/easily available or common, which I see as a big issue, as I don't seem to be able to just drop an order to AliExpress.
Obviously, If Espressif have some special relationship with the manufacturer I can understand, but if not, please focus on those most available and used for the different use cases. I cannot bet any business on that sensor I am afraid.

closed 2025-08-30 by nicklasb