espressif/esp-video-components #20: [Feature] Conversion to grayscale in PPA

mutatrum, 2025-02-10:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

QR decoders such as quirc like to get their image format in grayscale. This would be perfectly suited for the PPA, to increase processing efficiency. Maybe a more general color transformation operation would be an even better option, but not sure how feasible this would be. For a good UX, it would be preferable to have the source video stream in color, so the user can actually see the color picture while quirc is handling the QR decoding in the background on the greyscale image.

### Use cases

QR decoders, color manipulation or filtering of video stream, possibly more.

### Alternatives

Conversion of pixels to greyscale on the cpu. But having a PPA and not using it for something like this seems a shame.

mutatrum, 2025-02-10:
This was the most efficient version I could think of using the CPU, with a 64k lookup table for converting RGB565 to greyscale:

```c
static uint8_t rgb565_to_greyscale_lut[65536];

// Calculate scalar to go from 5 or 6 bit color to a single color channel luminance value
static float get_scalar(float luminance_coefficient, int color_bits)
{
    float max_value = (1 << color_bits) - 1;
    float scalar = 255 / max_value;
    return luminance_coefficient * scalar;
}

// Lookup table to convert uint16_t RGB565 to uint8_t greyscale
static void create_lut()
{
    const float SCALAR_RED = get_scalar(0.299, 5);
    const float SCALAR_GREEN = get_scalar(0.587, 6);
    const float SCALAR_BLUE = get_scalar(0.114, 5);

    // Lookup table index, RGB565 has 5 bits red, 6 bits green, 5 bits blue. By looping over the colors, the index location matches the RGB565 value.
    int i = 0;
    for (int red = 0; red < 1 << 5; red++) {
        for (int green = 0; green < 1 << 6; green++) {
            for (int blue = 0; blue < 1 << 5; blue++) {
                int grey = round(red * SCALAR_RED + green * SCALAR_GREEN + blue * SCALAR_BLUE);
                rgb565_to_greyscale_lut[i++] = grey;
            }
        }
    }
}

static void to_greyscale(uint16_t * b, uint8_t * dest, int width, int height)
{
    for (int i = 0; i < width * height; i++) {
        dest[i] = rgb565_to_greyscale_lut[b[i]];
    }
}
```

WangYuxin-esp, 2025-02-11:
Thank you for reporting the feature requirements. The [PPA peripheral](https://docs.espressif.com/projects/esp-idf/en/latest/esp32p4/api-reference/peripherals/ppa.html) currently does not support the conversion from RGB to Gray. We will consider discussing this feature in the next stage of design. For now, you can try:  
```
#ifndef DL_CLIP
#define DL_CLIP(x, low, high) ((x) < (low)) ? (low) : (((x) > (high)) ? (high) : (x))
#endif
inline uint8_t convert_pixel_rgb888_to_gray(int red, int green, int blue)
 {
            int temp = (red * 38 + green * 75 + blue * 15) >> 7;
            return DL_CLIP(temp, 0, 255);
 }
```

mutatrum, 2025-03-11:
I don't know how involved it would be, but it looks like adding a GREY color mode as the output color format of the SRM would be sufficient. This would also make it possible to do a scale down and color conversion in one operation, to do a faster QR detect on a lower resolution frame.

WangYuxin-esp, 2026-03-09:
> > [@mutatrum](https://github.com/mutatrum) The latest P4 chip(v3.1) now supports grayscale conversion and rotation、scale in PPA. Please apply for the latest P4 chip from the company website to test it.
> 
> So if I understand this coreecrly this is a new hardware revision of the P4?

Yes, absolutely correct.