espressif/esp-video-components #21: ESP32-P4 only?

nicklasb, 2025-02-24:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I expected this to work with the ESP32-S3, as the readme doesn't restrict its usage to any specific chip.

### Actual behavior (suspected bug)

However, the only target listed in idf_component.yml is esp32p4

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

Use the latest distributed package or the master branch.

### Project release version

0.9.0 / master branch

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

ZSH

donghengqaz, 2025-02-25:
Hi @nicklasb, in our plan, we will start to add the ESP32-S3 LCD_CAM DVP driver in the Q2(April ~ June), if you have some requirement about the time and software features, please let me know.

nicklasb, 2025-02-25:
Ok, I am afraid I cannot wait several months, since the readme states "ESP32 series chips", and it was from Espressif, I assumed that all chips was supported, you really need to be clearer in your communication. 
Is there no other way to use a DVP camera with a ESP32-S3?

donghengqaz, 2025-02-26:
@nicklasb Could you show me the link of the README about the inaccurate description, I will fix it. If you want to use DVP camera with ESP32-S3 currently, there is a ready-made project [esp32-camera](https://github.com/espressif/esp32-camera), but if this one can't match your requirement, I will try to add ESP32-S3 DVP driver and video device in March.