espressif/esp-video-components #23: Is it possible to change bitrate dynamically in esp_video?

OstapFerensovych, 2025-03-01:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hello,

I'm working with your esp_video library and would like to know if it's possible to change the bitrate dynamically while a video stream is running, or if I need to stop and restart the stream to apply new bitrate settings?

I've already tried changing qp_min, qp_max, and bitrate parameters, but couldn't see any visual difference in the output.

Thank you for your help!

houhaiyan, 2025-03-18:
@OstapFerensovych 
The esp_h264_enc_set_bitrate allows you to dynamically adjust bitrate.
Some suggestions:
Using hardware H264 coding, want to dynamically adjust the bit rate. RC mode must be used in the initial configuration. That is, qp_min and qp_max are not equal. The advice is a little different. If bitrate is configured, the bitrate will not be changed immediately. This response will slow down a bit, preventing the image from mutating, causing a mutation in the image quality.
Some doubts:
1. What are the qp_min and qp_max you tested?
What value bitrate tests.
2. How do you determine if it doesn't work?

donghengqaz, 2025-03-18:
@OstapFerensovych The changed configuration only works after restarting the video device, but I can try to add the real-time and dynamical feature for you.