espressif/esp-video-components #24: Stream buffer corruption with out of order release

mutatrum, 2025-03-14:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I have a multi-threaded system, where one thread is starting a v4l2 video stream with 3 buffers. This process handles display of the frames with LVGL, at about 15 fps. A second thread does QR decoding, at around 2 fps. As the QR decoding is a way slower process, I pass a frame from the display thread to the QR decode thread, and defer the frame release. When the second process is done with the QR decoding, it's send send back to the first process through another queue and the frame is release.

### Actual behavior (suspected bug)

Instead, I get frequent flickers of partial old frames. The idea is that with 3 buffers and one being used by the display task, one by the QR decoding, there is always a buffer available to accept a new frame from the stream.

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

Have a video stream, and occasionally wait a few 100ms with releasing a frame, while keeping the rest of the stream code running.

### Project release version

v0.8.0~1

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu 24.04

### Shell

Bash

mutatrum, 2025-03-14:
Video to show what's happening:

https://github.com/user-attachments/assets/053a717d-d7c8-4a93-87b5-d277c8d6cd72

If I disable the 2nd process, the video is smooth with only slight tearing, but no old frames.

mutatrum, 2025-03-14:
I'm trying to reproduce this based on one of the examples, but no luck as of yet.

donghengqaz, 2025-03-18:
The MIPI-CSI video device continues to receive the newest frame in the esp-video, do you use the DVP port to capture image?

mutatrum, 2025-03-19:
I've taken the video streaming code from https://github.com/espressif/esp-video-components/tree/master/esp_video/examples/simple_video_server, the /stream endpoint. I'll try to trim my code and post it here.

mutatrum, 2025-03-26:
To add, the camera is connected to the MIPI-CSI port.