espressif/esp-video-components #26: [question] esp_video uvc example

chegewara, 2025-03-20:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hi team,
i am testing uvc example from esp_video component and i have 2 observations:
- the video stream is delayed, the higher resolution the higher delay
- more important thing, when i setup USB uvc to use bulk instead of iso then on ubuntu and i try to open video with ffplay i see this message `ioctl(VIDIOC_STREAMON): Cannot allocate memory` 
memory issue is not related to frame size

I dont know if memory issue is on esp or ubuntu side.
Hardware:
- p4 devkit
- camera OV5647
- ubuntu 24.04, high end laptop

Thanks

chegewara, 2025-03-20:
Similar output is when i am trying to use VLC app:
```
cvlc v4l2:///dev/video2
VLC media player 3.0.20 Vetinari (revision 3.0.20-0-g6f0d0ab126b)
[00005fdc615361b0] dummy interface: using the dummy interface module...
libv4l2: error turning on stream: Cannot allocate memory
[00007f02dc001130] v4l2 demux error: cannot start streaming: Cannot allocate memory
[00007f02dc001130] v4l2 demux error: not a radio tuner device
libv4l2: error turning on stream: Input/output error
[00007f02dc00a630] v4l2 stream error: cannot start streaming: Input/output error
[00007f02e4000c90] main input error: Your input can't be opened
[00007f02e4000c90] main input error: VLC is unable to open the MRL 'v4l2:///dev/video2'. Check the log for details.
```
just in this case uvc example is crashing.

WangYuxin-esp, 2025-05-07:
This a USB communication problem. Try to configure the UVC data as JPEG data. Then try to open the MJPEG stream using [potplay](https://github.com/espressif/esp-video-components/tree/master/esp_video/examples/uvc#software-required) app.