espressif/esp-video-components #28: OV5640 support

Llgok, 2025-05-19:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

I would like to add MIPI support for the ov5640. [OV5640_CSP3_DS_2.01_Ruisipusheng](https://www.corecourse.cn/forum.php?mod=attachment&aid=NjI2OXxhYTVlYmQ0YnwxNzQ3NjM1Njc5fDB8MjkzMzE%3D)

I am using the ESP32-P4 to develop with the OV5640 camera module. I noticed that there is an existing OV5645 driver in the library. After comparing some registers, I found that the OV5645 and OV5640 are similar. Can I just modify a small part of the code to adapt it for the OV5640? The OV5640 is easier to purchase than the OV5645. Currently, although initialization was successful, the image has not been output. Here is my partial debugging log:
﻿
[2025-05-19 16:52:29.429] mipi dsi phy powered on
[2025-05-19 16:52:29.429] 
[2025-05-19 16:52:29.531] I (2060) ov5645: Detected Camera sensor PID=0x5640
[2025-05-19 16:52:29.531] 
[2025-05-19 16:52:29.704] I (2201) app_video: version: 0.9.0
[2025-05-19 16:52:29.704] I (2201) app_video: driver:  MIPI-CSI
[2025-05-19 16:52:29.704] I (2202) app_video: card:    MIPI-CSI
[2025-05-19 16:52:29.704] I (2202) app_video: bus:     esp32p4:MIPI-CSI
[2025-05-19 16:52:29.704] I (2204) app_video: width=1280 height=960
[2025-05-19 16:52:29.704] I (2208) app_video: Video Stream Start
[2025-05-19 16:52:29.704] 
[2025-05-19 16:52:29.736] E (2249) HAL_LOGE: mipi_dsi_host_ll_gen_is_cmd_fifo_full waiting
[2025-05-19 16:52:29.736] 
[2025-05-19 16:52:30.512] Discovered IIC devices[0]: 0x18
[2025-05-19 16:52:30.512] Discovered IIC devices[1]: 0x28
[2025-05-19 16:52:30.512] Discovered IIC devices[2]: 0x3c (ov5640)
[2025-05-19 16:52:30.512] Discovered IIC devices[3]: 0x58
[2025-05-19 16:52:30.512] Discovered IIC devices[4]: 0x68
    
﻿
The camera is a custom one, and it might be a problem with the camera itself. Do you have any suggestions?

### Use cases

https://github.com/espressif/esp-video-components/tree/master/esp_cam_sensor/sensors/ov5645

WangYuxin-esp, 2025-05-27:
Please try using the branch [here](https://github.com/WangYuxin-esp/esp-video-components/tree/feature/add_ov5640_mipi_csi_driver).
You can run the examples directly [here](https://github.com/WangYuxin-esp/esp-video-components/tree/feature/add_ov5640_mipi_csi_driver/esp_video/examples).

WangYuxin-esp, 2025-06-09:
@Llgok OV5640 can be used now.