espressif/esp-video-components #37: OV5645 displays an abnormality when using the YUV420 configuration.

lovely-him, 2025-09-05:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

- M2M requires the use of YUV420 data input, H.264 data output, and the camera captures images using MIPI_2lane_24Minput_YUV420_1280x960_30fps.
- (ESP32P4 IDF-V5.5.0)

### Actual behavior (suspected bug)

- The displayed H264 video image appears greenish.

[image]

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

- When using MIPI_2lane_24Minput_RGB565_1280x960_30fps and MIPI_2lane_24Minput_YUV422_1280x960_30fps, the MJPEG display is normal.

- Using PPA to convert RGB565 to YUV420 format for display works normally, so I think there is something wrong with this configuration item: MIPI_2lane_24Minput_YUV420_1280x960_30fps.

### Project release version

v1.3.0

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

windows 11

### Shell

ZSH

lovely-him, 2025-09-08:
- 另外发现两个问题, 根据手册, `ov5645` 的电压应该是 1900mv, `pwdn` 引脚的逻辑应该是和 `reset` 相同

```c
#define CSI_LDO_UNIT_ID                         (3)             // MIPI CSI LDO单元ID
#define CSI_LDO_CFG_VOL_MV                      (1900)          // MIPI CSI LDO配置电压（毫伏）根据正点原子手册描述需要修改这个

#define EXAMPLE_SCCB_I2C_INIT_BY_APP            (1)             // 是否由应用程序初始化SCCB I2C
#define EXAMPLE_SCCB_I2C_PORT_INIT_BY_APP       (I2C_NUM_0)     // SCCB I2C使用的端口号
#define EXAMPLE_SCCB_I2C_SDA_PIN_INIT_BY_APP    (33)            // SCCB I2C的SDA引脚号
#define EXAMPLE_SCCB_I2C_SCL_PIN_INIT_BY_APP    (32)            // SCCB I2C的SCL引脚号
#define EXAMPLE_MIPI_CSI_SCCB_I2C_FREQ          (100000)        // SCCB I2C通信频率(Hz)
#define EXAMPLE_MIPI_CSI_CAM_SENSOR_RESET_PIN   (34)            // 摄像头传感器复位引脚号(-1表示未使用)
#define EXAMPLE_MIPI_CSI_CAM_SENSOR_PWDN_PIN    (54)            // 摄像头传感器掉电引脚号(-1表示未使用)
#define EXAMPLE_MIPI_CSI_XCLK_PIN               (-1)            // 摄像头传感器外部时钟引脚号(-1表示未使用)
```

- 除非 `EXAMPLE_MIPI_CSI_CAM_SENSOR_PWDN_PIN` 宏定义设置为 `(-1)` , 否者一定要修改下面的引脚逻辑, 否者会无法通信, 找不到设备

```c
static esp_err_t ov5645_power_on(esp_cam_sensor_device_t *dev)
{
    esp_err_t ret = ESP_OK;

    if (dev->xclk_pin >= 0) {
        OV5645_ENABLE_OUT_XCLK(dev->xclk_pin, dev->xclk_freq_hz);
    }

    if (dev->pwdn_pin >= 0) {
        gpio_config_t conf = { 0 };
        conf.pin_bit_mask = 1LL << dev->pwdn_pin;
        conf.mode = GPIO_MODE_OUTPUT;
        ret = gpio_config(&conf);

        // carefully, logic is inverted compared to reset pin
        gpio_set_level(dev->pwdn_pin, 0);
        delay_ms(10);
        gpio_set_level(dev->pwdn_pin, 1);
        delay_ms(10);
    }

    if (dev->reset_pin >= 0) {
        gpio_config_t conf = { 0 };
        conf.pin_bit_mask = 1LL << dev->reset_pin;
        conf.mode = GPIO_MODE_OUTPUT;
        ret = gpio_config(&conf);

        gpio_set_level(dev->reset_pin, 0);
        delay_ms(10);
        gpio_set_level(dev->reset_pin, 1);
        delay_ms(10);
    }

    return ret;
}

static esp_err_t ov5645_power_off(esp_cam_sensor_device_t *dev)
{
    esp_err_t ret = ESP_OK;

    if (dev->xclk_pin >= 0) {
        OV5645_DISABLE_OUT_XCLK(dev->xclk_pin);
    }

    if (dev->pwdn_pin >= 0) {
        gpio_set_level(dev->pwdn_pin, 1);
        delay_ms(10);
        gpio_set_level(dev->pwdn_pin, 0);
        delay_ms(10);
    }

    if (dev->reset_pin >= 0) {
        gpio_set_level(dev->reset_pin, 1);
        delay_ms(10);
        gpio_set_level(dev->reset_pin, 0);
        delay_ms(10);
    }

    return ret;
}
```

```c
E (2647) i2c.master: I2C transaction unexpected nack detected
E (2648) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
E (2659) i2c.master: i2c_master_transmit_receive(1248): I2C transaction failed
E (2660) sccb_i2c: s_sccb_i2c_transmit_receive_reg_a16v8(116): faled to transmit receive
E (2671) ov5645: ov5645_get_sensor_id(214): read pid_h failed
E (2682) ov5645: Camera get sensor ID failed
E (2722) esp_video_init: failed to detect MIPI-CSI camera sensor with address=3c
```

- 摄像头手册有如下描述, 确保符合才能工作

```
note1 : PWDNB, active low, should be controlled on the system level
```

```
2.7 power up sequence
Based on the system power configuration (1.8V or 2.8V for I/O power, using external DVDD or internal DVDD, requiring 
access to the SCCB during power up period or not), the power up sequence will differ. If 1.8V is used for I/O power, using 
the internal DVDD is preferred. If 2.8V is used for I/O power, due to a high voltage drop in the internal DVDD regulator, 
there is a potential heating issue. Hence, for a 2.8V power system, OmniVision recommends using an external DVDD 
power supply. Due to the higher power down current when using external DVDD, OmniVision strongly recommends 
cutting off all power, including the external DVDD, when the sensor is not in use in the case of 2.8V I/O and external 
DVDD.
2.7.1 power up with internal DVDD
For powering up with an internal DVDD and SCCB access during the power ON period, the following conditions must 
occur:
1. when DOVDD and AVDD are turned ON, make sure DOVDD becomes stable before AVDD becomes stable
2. PWDNB is active low with an asynchronized design (does not need clock)
3. PWDNB must be low if SCCB is accessed during the power up period
4. if PWDNB pin is controlled as below, for PWDNB to go high, power must first become stable. System needs 
to drive PWNDB to high as the sensor has an internal pull down on PWDNB pin (AVDD to PWDNB ≥ 5 ms)
5. RESETB is active low with an asynchronized design
6. master clock XVCLK should be provided at least 1 ms before host accesses the sensor’s registers
7. host can access SCCB bus (if shared) during entire period. 20ms after RESETB goes high, host can access 
the sensor's registers to initialize sensor
```

WangYuxin-esp, 2025-09-09:
Thanks for reporting this issue.  
1）对于 OV5645 output YUV420 + H.264 encoder 一起使用时，出现颜色问题，我们已经知道了。本质原因是 sensor 输出的 YUV sequence 与 H.264 encoder 需要的 sequence 不一致。我们已经在最新的芯片上允许 H.264 支持输入 YUV422 等格式。  
2）对于 power on sequence 和 power off sequence，我们这里仅提供了一个参考写法，用户可以对 xclk pin、reset pin、pwdn pin 在应用层代码中通过 gpio 来控制。因为 sensor 上电的管脚 DVDD、DOVDD、AVDD 的设计是灵活的，可能和 SOC 一起上电，也可能有其他方式，这总是和硬件设计相关的，驱动难以预先知道连接的模组到底是如何设计的。当然，也可以直接更改这里的 power_on()、power_off() 函数使之与你的硬件设计匹配。

lovely-him, 2025-09-09:
> Thanks for reporting this issue. 1）对于 OV5645 output YUV420 + H.264 encoder 一起使用时，出现颜色问题，我们已经知道了。本质原因是 sensor 输出的 YUV sequence 与 H.264 encoder 需要的 sequence 不一致。我们已经在最新的芯片上允许 H.264 支持输入 YUV422 等格式。 [...]

Thank you for your response.
感谢您的回答.

请问我应该如何在 `H.264` 上使用 `YUV422` 等格式呢, 是需要更新 `ESP32P4` 的固件吗? 还是需要更新 `esp-idf` 或 `esp-video` ? 
期待您的回复.

WangYuxin-esp, 2025-09-09:
> Thank you for your response. 感谢您的回答.
>
> 请问我应该如何在 `H.264` 上使用 `YUV422` 等格式呢, 是需要更新 `ESP32P4` 的固件吗? 还是需要更新 `esp-idf` 或 `esp-video` ? 期待您的回复.

是需要新版本的芯片。虽然都叫做 P4，但是它有修复版。如果是新的商用项目，建议联系商务，申请新的芯片进行测试。  
当然，也可以尝试 OV2710 这类 sensor，它经过 P4 ISP 处理后的 YUV420 数据可以直接用于 H.264 编码，不存在 data seq 问题。

closed 2025-10-23 by lovely-him

Devin-Kung, 2025-11-19:
> SW encoder 的兼容范围广一些，但是带宽能力肯定不如 HW encoder。 HW H.264 推荐的用法，是用一个 RAW sensor，比如 OV2710\SC2336，经过 ISP 的处理和转换，输出 YUV420，然后给 HW H.264 使用。 HW H.264 的使用包含两种，在 DQBUF 获取图像数据后，直接使用 driver API，进行 process()。或者使用 [M2M 机制](https://github.com/espressif/esp-video-components/blob/master/esp_video/examples/uvc/main/uvc_example.c#L343)。

感谢回复。

我们已经尝试过SC2336了，但是画面效果不是很满意，自己调整ISP参数有一定难度，输出的效果也是不如OV5645这种自带ISP模块的摄像头；另外，使用OV5645是考虑到有自动对焦功能(AF)，后面我们也准备试试OV2710。

就是说最新的芯片版本，硬件编码也是只支持ESP_H264_RAW_FMT_O_UYY_E_VYY这种格式是吗？

houhaiyan, 2025-11-19:
@Devin-Kung 最新的硬件版本支持如下：

[image]
最新的版本计划本周进行发布

Devin-Kung, 2025-11-19:
> [@Devin-Kung](https://github.com/Devin-Kung) 最新的硬件版本支持如下：
> 
> <img alt="Image" width="857" height="438" src="https://private-user-images.githubusercontent.com/56857816/516111982-a912c2fc-1dba-4aac-919b-4eb6e8754a08.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjM1MzY2OTUsIm5iZiI6MTc2MzUzNjM5NSwicGF0aCI6Ii81Njg1NzgxNi81MTYxMTE5ODItYTkxMmMyZmMtMWRiYS00YWFjLTkxOWItNGViNmU4NzU0YTA4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTExMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUxMTE5VDA3MTMxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI2NDVhMzRhMDE1MWU5OTYwZWQ0ZmU1N2FjMDZiZjEzYjgzY2Y4Y2E4NWM1MDQxYjZmOWRjN2E1OGJhMDFlZjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Kedylw6KWqXZCURvU1t4N9ORFKr10sZmxFOEBFBPnt0"> 最新的版本计划本周进行发布

@houhaiyan 

感谢您的回复。

新增的这些格式已经可以满足我们的需求了，如果要使用图片里新增的格式进行硬件编码的话，是否也需要使用最新版的芯片；
最新版本的芯片的版本号批次号您这边有吗，我们计划采购一些，或者您有采购的联系方式吗？

houhaiyan, 2025-11-25:
@Devin-Kung 新的H264组件 已经更新。 新的格式支持需要新的芯片。 如果需要采购 https://www.espressif.com/zh-hans/contact-us/sales-questions  联系我司 sales.

Devin-Kung, 2025-11-25:
> [@Devin-Kung](https://github.com/Devin-Kung) 新的H264组件 已经更新。 新的格式支持需要新的芯片。 如果需要采购 https://www.espressif.com/zh-hans/contact-us/sales-questions 联系我司 sales.

了解了，多谢！！