espressif/esp-video-components #41: cam sensor component not compatible with idf5.5+

chegewara, 2025-09-29:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

To build with esp-idf 5.5.0+, or at least to pass CI.

### Actual behavior (suspected bug)

This is code where it fails to build (maybe more code too)
https://github.com/espressif/esp-video-components/blob/master/esp_cam_sensor/src/driver_spi/spi_slave.c#L61-L77

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

idf.py add-dependency "espressif/esp_video^1.2.0"

build

### Project release version

esp-video 1.2.0+

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

ubuntu

### Shell

Bash

chegewara, 2025-09-29:
https://github.com/espressif/esp-video-components/blob/master/esp_video/src/data_reprocessing/esp32p4/esp_video_swap_byte.c#L93

`bitscrambler_disable` - does not exist

donghengqaz, 2025-09-30:
Hi @chegewara , I check your commit ID, this is a very older version between v5.4 and v5.5, you can the [v5.5](https://github.com/espressif/esp-idf/releases/tag/v5.5) was released in Jul 21. Actually the code management solution only can manage the code after some released tag, but can't manage the code before the released tag, especially some older commit ID.